IBIResearch/RedPitayaDAQServer #2: Single image for both master and slave

jeanminet, 2017-10-04:
Hello,

I was wondering if it was possible to use the same bitstream for both master and slave.
Would this connection work ? :

[image]

Best,
Jean

tknopp, 2017-10-04:
Hi Jean, it would be really great if this would be possible. I just know that Jonas tried something which did not work. He meant that the issue is, that the slave has no clock at that time when this decision was made.
I cannot comment on the FPGA stuff since I only did the entire C stuff. Ping @jbeuke therefore.

jbeuke, 2017-10-04:
@jeanminet My first idea was to expose the master/slave settings to the cpu via a configuration register. This did not work, since I need a clock for the register which I don't get, if the clock selection is wrong. Your idea is actually a really nice one since the locked output of clk_wiz_1 should only be high if a clock is supplied via the daisy chain connector. I will add this to the blockdesign and have @tknopp test it in hardware ;)

tknopp, 2017-10-04:
yes sure, I can test. You could make a branch that I can test.

jbeuke, 2017-10-23:
Apparently, this does not work as expected. The acquisition pointer sometimes stops or jumps, which makes the acquisition unreliable. Since using two different bitfiles for master and slave is not a major issue, I postpone digging further for a solution. Help is appreciated though!

mgraeser, 2020-04-05:
I included the following solution to the branch v2:
[image]
The select is driven bei ext_DIO0_N. On the other side of the BlockDiagram ext_DIO1_N is given a constant 0. Now the DIO0_N is physically between a high (3.3V on External conn) and low (DIO1_N).
With a jumper between DIO1_N and DIO_0_N the device is set as Master. Between 3.3V and DIO0_N it is set to slave. 
The advantage here is that the DIOs do not need the clock to work.

closed 2020-04-10 by tknopp