IBIResearch/RedPitayaDAQServer #5: Correct the amplitude response of the ADC

jbeuke, 2017-10-16:
The CIC-Compiler has a non-flat amplitude response. This can be compensated according to [this article](https://www.koheron.com/blog/2016/10/03/decimator-cic-filter). Additionally the input stage has some issues according to [this forum entry](http://forum.redpitaya.com/viewtopic.php?f=9&t=468). The original image includes a [compensation filter](https://github.com/RedPitaya/red-pitaya-notes/blob/master/projects/red_pitaya/red_pitaya_dfilt1.v) which uses the values from [this forum entry](http://forum.redpitaya.com/viewtopic.php?f=14&t=1473&p=5406#p5406). Ideally both filters should be combined in order to achieve a minimal delay.

amarrei, 2021-04-09:
Hello,
I have measured the ADC inputs to calculate offset and gain correction to be done on the client side. My values are similar to the ones Tobias mentioned in another issue, so this looked promising.
This works for DC and very low frequencies, but already with a few 10s of kHz, I see an attenuation when plotting the acquisition.
After undoing my HW modifications I can see that this is not coming from my additions. I haven't yet done further measurements, but the attenuation is not negligible.
I am not sure if this is related to this dfilt1 filter or not. New to FPGA, I have not yet figured out where/how it is actually implemented in this project.
Has this issue been resolved and I have a different problem?
Thankful for any help,
Amir

tknopp, 2021-04-10:
no, we have not resolved this issue.

jonschumacher, 2023-02-07:
@nHackel @jusack Is this resolved now?

nHackel, 2023-03-17:
Yes, the implementation is finished and will be included in the upcoming release. Only the documentation of the FIR filter is missing and will be added later

jonschumacher, 2023-09-28:
@jusack noticed that there is still an issue with the input stage when trying to capture square waves. This might still be related to the missing compensation filter. The link from the original post is broken but the file can be found here: https://github.com/RedPitaya/RedPitaya-FPGA/blob/e5f70e1dee2c633eab5c162cca25a085d10c7b55/rtl/red_pitaya_dfilt1.sv

jonschumacher, 2024-08-01:
@jusack Is this done?

closed 2024-12-08 by jonschumacher