IBIResearch/RedPitayaDAQServer #6: Improve the build system

jbeuke, 2017-10-17:
The build system is currently split into several Makefiles which are invoked by a root folder Makefile and contains absolute paths to libraries. This should be centralized and be made more flexible. A bonus might be to provide different targets for building and installing the server application aswell as the FPGA image.

tknopp, 2017-10-17:
yes, although this usually is the point where one should ask if `make` is really the right tool for the job. There are modern build systems that are much easier to use.

tknopp, 2017-10-17:
`cmake` is ok. `qmake` was from perspective a little nicer. Nowadays `Meson` http://mesonbuild.com/index.html seems to get popular. The syntax looks pretty nice.

We jut have to look if these are available in the repository of the debian that is installed on the RP.

jbeuke, 2017-10-17:
Not only the Debian but also the Alpine Linux which Pavel Demin is now using. But this is the case: https://pkgs.alpinelinux.org/package/edge/main/armhf/meson

But using a different build tool sounds good. I don't really get a grasp on `make`. But thats's something we can do when the rest is working nicely.

jbeuke, 2020-04-24:
I have just set up a RP completely from scratch. `make` worked flawlessly and I think the FPGA build process now works as well. Can we close this?

tknopp, 2020-04-24:
I think at this point I don't want so switch the build tools and am fine with the current situation. So yes this can be closed for now.

closed 2020-04-24 by tknopp