IBIResearch/RedPitayaDAQServer #7: Add selection for HV/LV

jbeuke, 2017-10-20:
This issue is related to #5: The input stage has some flaws which are being corrected by a filter. This filter has two constants AA and BB which differ for LV and HV.

tknopp, 2017-10-20:
I also want to store the mapping IntToVolt on the RP. The server should provide a calibration procedure for this. Here, LV and HV need to be selected

tknopp, 2017-10-23:
We should use the eeprom stuff that is implemented in the regular RP API:

https://github.com/RedPitaya/RedPitaya/blob/1dfabae67968cc76548e461c7d123a284d3df725/api/src/calib.c

If that is done the DAC API should use voltage.

jongra, 2021-05-06:
Is there a working solution for the selection today?

To be honest, I didn't realy understand how to do the calibration you mentioned.

jonschumacher, 2021-05-06:
Unfortunately not yet. We do not use the „high“ voltage input path. I would recommend to just do a test sweep in the relevant frequency range with a calibrated signal generator and compare the acquired amplitude with the setting from the signal generator. If this comparison is fine for you, this issue might not even be relevant for you. Or did you already experience problems?

jongra, 2021-05-07:
Thank you for your answer.
In general, using a signal generator as a reference would work for us.
My problem is more that I have to use the 1V scale (LV mode) for finer voltage resolution and smaller absolute noise.

tknopp, 2021-05-07:
So there is two different issues here:
1. We have a transfer function in the input due to the decimation. This implies that higher frequencies are damped.
2. Our interface currently only provides integers.

What we do regarding 2. is shown here: https://github.com/MagneticParticleImaging/MPIMeasurements.jl/blob/master/src/DAQ/RedPitayaScpiNew.jl#L160  
So we have two calibration values (scaling and offset) that we use to convert int to volt. Roughly the values are:
   scaling = 0.00012957305   offset = 0.015548877
for LV.

jongra, 2021-05-07:
Sorry, but for 1 I don't understand what you mean with "transfer function".

Regarding number 2, I don't really care if the data format is in integers or floats. I guess the conversion can be done later on in the software.
As far as I understand from the [redpitaya schematics](https://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/shem.html), a jumper decides the coupling of the voltage input to a fixed amplifier and further to the fixed ADC. So, I'll get a better voltage resolution when I'm using the LV jumper position. The software part than has to adjust the numerical scaling of the ADC output - I think this is what you suggested. However, you referred to a completely different repository. Where can I find the scaling in the RedPitayaDAQServer project?

jonschumacher, 2021-05-07:
Concerning the transfer function I would like to refer to [Wikipedia](https://en.m.wikipedia.org/wiki/Transfer_function). The DAQ project just outputs the raw data of the 14 Bit ADC. The full scale value of the ADC is manipulated with said jumper.
 The missing scaling is exactly what is meant within this issue and is an open problem which is solved indirectly in the referenced package.