IBIResearch/RedPitayaDAQServer #11: Write Pointer Sometimes Set to zero

tknopp, 2017-11-02:
I have sometimes the problem that the write pointer of the ADC:
```
	adc_sts = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, mmapfd, 0x40001000);
```
remains zero. This issue is here to track this. @jbeuke: Is it correct that the write pointer does always run or do we need to manually start it? 
Is it correct that this is the FPGA code from @pavel-demin? Or have you modified it?

The strange thing is that one has to reboot the RP when this happens. Reloading the the image does not change anything.

jbeuke, 2017-11-02:
The write pointer runs continuously and without the need for a manual start. It is only reset to zero if the core gets a reset signal. This happens when the ADC mode is changed to 'triggered' and the master trigger is switched on and off. Since this normally works without problems and reflashing the FPGA image resets all signals I suspect problems with the bus. As far as I remember I did not change the code which writes the data to the RAM. I get the same problems but until now I was unable to isolate the conditions for this situation.

tknopp, 2017-11-02:
Thanks Jonas. Then its good to have this issue to track all information about this. Maybe @pavel-demin has an idea?

jbeuke, 2017-11-03:
The first thing I suspect might be my patch for an opt_design error in the status register core that was also included in red-pitaya-notes: https://github.com/pavel-demin/red-pitaya-notes/blob/42099f4fa3912552900a4b4c2a5621033125703c/cores/axi_sts_register_v1_0/axi_sts_register.v#L96

Having the wrong signals on the AXI bus might cause the problems. But since I am not very familiar with the internals of AXI, I am unable to neither decide if this is really the case nor implement the correct signals at the moment.

pavel-demin, 2017-11-03:
> The write pointer runs continuously and without the need for a manual start. It is only reset to zero if the core gets a reset signal.

If a burst write transaction is interrupted, then the AXI bus hangs. It's a known feature of the ZYNQ chips.

Before resetting axis_ram_writer, the stream of the incoming data should be stopped and an ongoing burst write transaction should be properly terminated.

jbeuke, 2017-11-03:
Do you think this might be done by delaying the reset signal until the reset for the previous cores has taken place? Or is there a nicer way of doing this?

pavel-demin, 2017-11-03:
> Do you think this might be done by delaying the reset signal until the reset for the previous cores has taken place?

Reading and writing to the sts registers is quite slow, so the following should be enough:
```
// reset previous cores
*rst &= ~1;
// reset ram writer
*rst &= ~2;
```

> Or is there a nicer way of doing this?

You could try to check the status of int_empty_wire:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_ram_writer_v1_0/axis_ram_writer.v#L70

jbeuke, 2017-11-04:
Okay, so I guess this could be fixed by changing [this](https://github.com/pavel-demin/red-pitaya-notes/blob/42099f4fa3912552900a4b4c2a5621033125703c/cores/axis_ram_writer_v1_0/axis_ram_writer.v#L82) line to `if(~aresetn && int_empty_wire)` and [this](https://github.com/pavel-demin/red-pitaya-notes/blob/42099f4fa3912552900a4b4c2a5621033125703c/cores/axis_ram_writer_v1_0/axis_ram_writer.v#L71) accordingly. But why do you use ALMOSTEMPTY instead of EMPTY for int_empty_wire?

tknopp, 2017-11-04:
@jbeuke: We could add the following test. Write a short C program that regularly performs the reset. Sooner or later the write Pointer will stay 0.

pavel-demin, 2017-11-04:
> But why do you use ALMOSTEMPTY instead of EMPTY for int_empty_wire?

To have at least 16 entries in FIFO before starting a burst write transaction.

tknopp, 2017-11-04:
Will the different RP actually be sample synchronous if they have to wait for an empty wire?

tknopp, 2017-11-04:
(just in case this is not clear: we use an external trigger to make different RP synchronous)

pavel-demin, 2017-11-04:
> Will the different RP actually be sample synchronous if they have to wait for an empty wire?

It depends more on how you reset all other modules.

The samples should be synchronous at the axis_ram_writer input, and if the axis_ram_writer FIFO buffer should be emptied before starting the data acquisition.

jbeuke, 2017-11-04:
Since this does only affect the shutdown of the transaction and not the startup, I reckon this does not interfere with the synchronicity of the devices. The FIFO should be empty on startup and therefore the reset change would be asserted as soon as possible. I will change this and we will see if it works.

tknopp, 2017-11-04:
sounds very good. Solving this will be a great step forward since the recent debugging sessions that we had on other issues where also effected by this one.

tknopp, 2017-11-05:
@jbeuke: I added a test program to reproduce the issue:

https://github.com/tknopp/RedPitayaDAQServer/blob/master/src/test/test_trigger.c

Here is a sample run:

```
root@red-pitaya:~# /root/RedPitayaDAQServer/build/test/test_trigger 
Decimation = 16 
____ 864 0 864 
____ 800 0 800 
____ 800 0 800 
____ 800 0 800 
____ 800 0 800 
____ 800 0 800 
____ 800 0 800 
____ 328 0 328 
____ 0 0 0 
Write Pointer remains zero!
```

you can see: 8 times the reset was successful. Then the AXI bus hangs. You have to reboot the RP in order to run the it again.

jbeuke, 2017-11-06:
It seems like this fixes the issue with resets, but this is not the case with reloading the bitfile. I guess this works if we make sure, that the trigger is disabled prior to reloading the image. @tknopp Can you confirm, that this is now working?

tknopp, 2017-11-06:
will try.

tknopp, 2017-11-06:
I have tried it and it seems to be fixed. The second time hangs again (as you expected due to reload of image)

In the server we could switch the trigger of once the acq has started. Would this solve the problem?

jbeuke, 2017-11-06:
Turning the trigger off would result in a stop of the acquisition. I guess reloading the bitfile everytime the server starts might be a problem. But I don't know a good way to check if the bitfile has already been loaded.

tknopp, 2017-11-07:
If we concentrate on the scpi server, we actually have that situation. The scpi server is started at startup and should in a regular use case not crash. Clients have to properly cleanup, so that the server is always ready for a new measurement.

jbeuke, 2017-11-23:
Since the problem seems to be gone, I will close this issue for now. Even stopping the server by Ctrl+C during an acquisition did not trigger the problem.

tknopp, 2018-01-05:
@jbeuke: I implemented logic that the system image is only loaded once

   https://github.com/tknopp/RedPitayaDAQServer/blob/master/src/lib/rp-daq-lib.c#L41

It relies on the fact that usually the `/tmp` directory is cleaned. Could you check if this is the case for Alpine linux?

tknopp, 2018-03-26:
@jbeuke this is unfortunately back again. I think I can even reproduce it. But currently with all our software stack.

tknopp, 2018-03-26:
I can reproduce the bug and have updated the "test_trigger_script" accordingly. @jbeuke or @pavel-demin. Would be great if you would have an idea why 
  https://github.com/tknopp/RedPitayaDAQServer/commit/67b7855b9ce6f0d93e89b730677d9ca020432b30

did not fix ist.

tknopp, 2018-03-26:
seems to be related:
https://forums.xilinx.com/t5/Embedded-Processor-System-Design/AXI-HP-Interface-Reset/td-p/664480

jbeuke, 2018-04-09:
Maybe it is possible to delay the reset until the buffer has been written. Since no further data should be coming due to the reset of previous cores, this should be enough to reset the DMA core when it is in a defined state. I hope to try this in the next days.

tknopp, 2018-04-09:
This would be really great if you could test it. My other workaround would be to only reset once and not for each data acquisition. But this would mean a total restructuring of the server code. When this bug is fixes the RedPitaya is a pretty reliable platform for our purpose.

tknopp, 2020-03-15:
@pavel-demin: I am currently working on fixing the bug that the axi bus hangs in the `axis_ram_writer` and therefore try to understand what @jbeuke and you have been communicating within this thread. First: Jonas implemented a solution in https://github.com/tknopp/RedPitayaDAQServer/commit/67b7855b9ce6f0d93e89b730677d9ca020432b30 which partly solved the hang. However, it also leads to the situation that the write pointer is not set to zero sometimes, which of course is an issue.

Your original advice was to flip a bit on `rst` but it is not clear to me what that is. Is that the config register of the `axis_ram_writer`? Or is that another config register?

What is important to mention is that `aresetn` is connected to an DIO pin so that we can trigger multiple RP simultaneously. I have, however, control to set a register before applying the trigger.

pavel-demin, 2020-03-16:
> Your original advice was to flip a bit on rst but it is not clear to me what that is. Is that the config register of the axis_ram_writer? Or is that another config register?

I'm afraid that you're misinterpreting my original advice. I've just checked that the full version of my original advice is still in [my first comment to this issue.](https://github.com/tknopp/RedPitayaDAQServer/issues/11#issuecomment-341647730)

The code with `rst` in my second comment was an abstract example to illustrate my answer to an additional question about delays. Here is an even more abstract version of this example without using the `rst` variable:
```
// reset previous cores
code resetting/stopping previous cores;
// reset ram writer
code resetting/stopping ram writer;
```

tknopp, 2020-03-16:
Ok, thanks so the block design looks as follows:

[image]
What I would need to do is to stop the data flow. Is there some general mechanism for that? In that example it would be best to stop the data in `util_vector_logic_0` but its not clear for me howto do that.

pavel-demin, 2020-03-16:
The `aresetn` port of `axis_ram_writer_1` must be controlled independently of the `aresetn` ports of the other blocks. In your screenshot, they are all connected together. This should be fixed.

Once it's fixed, the data flow can be stopped by putting the other blocks to their reset state. It can be done by setting their `aresetn` to 0.

tknopp, 2020-03-16:
Ah, ok, if I understand you correctly, the issue is only at shutdown of the module. Might it be an option to delay the reset? so that I put something in-between the `aresetn` port of the ram writer and that delays if the signal goes from 1 to 0 but has no delay the other way around?

pavel-demin, 2020-03-16:
> Ah, ok, if I understand you correctly, the issue is only at shutdown of the module.

Since you never explained how to reproduce this problem, I don't know what exactly doesn't work in your setup. 

Reading the comments to this issue, I'd suspect one of the following cases:
 - you're re-configuring the FPGA while the DMA transfers are running
 - you're resetting the RAM writer module while the DMA transfers are running

I think that in both cases the solution is to stop the DMA transfers before re-configuring the FPGA or before resetting the RAM writer module.

Since I don't think that I can contribute anything else to the resolution of this issue, I unsubscribe.

mgraeser, 2020-04-07:
We have now solved the issue in branch v2. The solution was very similar to what Pavel suggested. Now the RamWriter aresetn is connected with the bit 0 of the cfg register. As soon as the server sets the Master trigger to low the datastream to the RamWriter comes to a stop. The server now monitors the write pointer. As soon as it stops it sets the cfg bit 0 to 0 and resets the RamWriter.

https://github.com/tknopp/RedPitayaDAQServer/blob/ec39ad4f695eaa61911b4ea260584272ccd96d57/src/lib/rp-daq-lib.c#L726

With this the DAC and ADC are synchronous, and the RamWriter does not hang.

tknopp, 2020-04-10:
this issue is resolved

closed 2020-04-10 by tknopp