IBIResearch/RedPitayaDAQServer #12: Buffered acquisition of slow ADC

jbeuke, 2017-12-04:
Until now, the synchronous acquisition of the slow ADC values is not possible since they are neither buffered nor timestamped. In order to use them properly, this is an important feature. This thread is for discussing possible solutions.

tknopp, 2017-12-04:
yep this would be awesome if we could get this feature working on the FPGA. For me it would be important that the frequency is an exact multiple of 125MHz/decimation of the fast ADC.

Then one could upload a LUT and the values would be updated periodically.

jbeuke, 2017-12-04:
Where would you need a LUT with the ADC? Or do you mean a FIFO/ringbuffer?

tknopp, 2017-12-04:
A ring buffer needs to be feed by values. We would need a thread on the CPU for that. Better is that you upload the pattern that should be cyclically applied.

tknopp, 2017-12-04:
Look here: https://github.com/RedPitaya/RedPitaya/blob/62238d4bf13a67c309ab008e942249a9146a2750/Examples/C/generate_arbitrary_waveform.c

This is the same concept. (just for the fast DAC)

jbeuke, 2017-12-04:
Well, this is true for the slow DAC but not for the slow ADC/XADC.

tknopp, 2017-12-04:
Ahh, I see, sorry for confusing things. Proposal:
 1. Slow ADC: buffer that is periodically filled
 2. Slow DAC: arbitrary waveform buffer is periodically applied.

Does this sound good?

For 2. we would have to relate the sampling rate to the fast components.

jbeuke, 2020-04-24:
As far as I can see it this is now a thing, right?
https://github.com/tknopp/RedPitayaDAQServer.jl/blob/e3f1f153fa02a53f43f750f69f819889fba50dd1/src/ADC.jl#L182

mgraeser, 2020-04-24:
Yes Slow DAC and FastDAC are now synchronous. SlowDac is buffered with a register

jbeuke, 2020-04-24:
Wait, I thought we are talking about the slow ADC here. But this seems to be not completely implemented yet, since [here](https://github.com/tknopp/RedPitayaDAQServer/blob/1d8813ad070a414df59b129dec432ae630b311ac/src/server/scpi_commands.c#L601) it is commented out.

mgraeser, 2020-04-24:
Ah sorry. ADC I have not changed yet

tknopp, 2020-04-24:
Yes, this issue is still open. In Principle we should be able to implement this but we really need to use something like the BRAM and buffer the values there. My feeling is that it should be possible to reach a sampling of about 25kHz with the slow ADCs. What would happen is the the control thread (previously named slow DAC thread) would then not only be responsible for feeding the slow DAC buffer but also the slow ADC buffer.

Implementation wise this all seems not to be so complicated. Just like our new PDM but reversed. Both the PDM and the XADC should either use the BRAM or direct memory. In this post:
  http://antonpotocnik.com/?p=514765
Anton talks about the BRAM and how to use that (presumedly he uses the code from Pavel) and in that post he also mentions 
```If we need more measurement storage we can use Red Pitaya’s on-board 512 MB DDR memory. We will show how to do that in future projects.```
So that is something we need to ask him. Maybe we should directly skip the BRAM implementation.