IBIResearch/RedPitayaDAQServer #15: Chose a single Linux Image

tknopp, 2018-01-05:
I currently still use the Debian image that cannot be downloaded anymore. We could switch to Alpine, although we have to clarify the bootup things there.

Another option would be the regular RP image. Last time I tried it, the system freezed. Regular RP image has the advantage that one may still use the web apps that can be handy from time to time.

@jbeuke: What is you opinion?

jbeuke, 2018-01-15:
While Alpine seems to be the perfect fit due its minimalistic approach, the standard image should be the way to go since the continuous support for it is more likely than that for Alpine. I never tried to run the server on the standard image, but if it works, I definitely prefer it this way. I guess for running the server it is necessary to kill at least the webserver and maybe other processes which draw network bandwidth and/or processor time.

tknopp, 2018-01-15:
I tried it last week and had some issues. But I don't remember what :-)

So lets keep that as a TODO. Alpine works now, also with automatic startup.

(you don't need to kill processes on standard image. Just deselect the services)

jbeuke, 2020-04-24:
Looks like Alpine is the choice to go ;)

closed 2020-04-24 by jbeuke

tknopp, 2020-04-24:
yes, because I increased the free RAM for the RamWriter  it is currently not safe to use any other Linux. One would need to incorporate that.