IBIResearch/RedPitayaDAQServer #16: Executing on single RP board

falcon98, 2018-08-07:
Hi,

I want to use your program for combining slow adc acquisition and fast adc acquisition(event based).
Can you please guide how to test your program since there are multiple bit files. How can I execute or modify the code for running an application for a single RP board in order to acquire both slow and fast adc data. The slow adc need to be continuous and fast adc data is needed on occurence of events.
Thanks a lot for your support

Regards,
David Falcon

tknopp, 2018-08-07:
This is possible but needs various modifications to the C server code. The Server ist build around the requirement so sample the fast ADC continuously. 

Sampling the slow adc is somewhat tricky since there is no buffering. What ist the sampling frequency YouTube require?

falcon98, 2018-08-07:
I am targeting 1 KSps for slow ADCs. 
In my application I need to acquire long duration (1-2 hrs) data from optical signals and need high sampling only for certain events.
Hence I thought of modifying your code for long duration logging and using fast ADCs at 100 MSps for 1 ms event durations.

Pls let me know if you need more information.

jbeuke, 2018-08-08:
What kind of event triggers your fast ADC? Does the acquisition from the slow ADC run continuously during the fast acquisition?

The problem with simultaneously acquiring from slow and fast ADC is the bandwidth of the connection from FPGA to RAM. Do you need both fast ADCs? If not, you could could replace one fast ADC channel with the slow ADC (maybe filling up with either repeated samples or zeros). Or you could simply use the fast ADC...

falcon98, 2018-08-08:
For event trigger, simple TTL pulse on GPIO can be used for trigger.

Yes acquisition from slow ADC is needed. Oly single Fast ADC is required.

Are you suggesting that I average the samples to arrive at lower sampling rates and disable averaging as event occurs?
How can i start to achieve the objectives?
Thanks

jbeuke, 2018-08-08:
Well, do you really need to trigger it? I guess the easiest way would be to disable one of the fast ADC in the FPGA image and plug in the slow ADC instead. Then you just use the server code for the fast acquisition and throw away all the garbage in between. All other trigger approaches would also require changes to the FPGA, since we currently just have a common trigger that distributes a "start fast acquisition" signal to all RPs.

falcon98, 2018-08-08:
Thanks a lot for your reply.
As per my understanding I will build the project as mentioned in your wiki page.
Run the daq_server.c code at RP board and execute python code on my windows laptop.
Are these steps correct? I will update here after testing.

Thanks a lot

jbeuke, 2018-08-08:
Sounds right. If you don't need to change the FPGA, you can simply install Alpine and then build the server. If there are problems with the Python code, maybe also check the MATLAB and Julia code, since the Python stuff has been tested less often.

tknopp, 2018-08-08:
I think the instructions are old. Nowadays we use the scpi server.

In general I wonder if the last would not be easier to be solved using the regular RP image. @falcon98 
do you really need 125 MS/s. This package will only give you 1/8 of that.

falcon98, 2018-08-08:
THanks for the reply.
I understand that transmission of data at 125 MSPs will not be possible due to network issues. My signal BW is 1-2 MHz so 125/8 will work.
As suggested I will build the daq_server_scpi.c file at RP for the testing. I hope the vivado project can be built by the given instructions.
I am sorry but with so many files in the code, it is a bit confusing for me, that why I asked here. After testing I will post an update here.

Thanks for help.

jbeuke, 2018-08-18:
Hi @falcon98 
Have you been successful in building and running the code?

jbeuke, 2018-09-04:
This seems to be either resolved or not of interest anymore -> Closed

closed 2018-09-04 by jbeuke