IBIResearch/RedPitayaDAQServer #17: Add signal generation in rasterized mode

jbeuke, 2018-09-12:
Currently, other signal types than sine can only be used when using the standard mode for the DDS compilers. It would be nice to have this working when in rasterized mode.

tknopp, 2018-09-12:
yes. Is there some limitation how low we can make the frequency here?

jbeuke, 2018-09-12:
Since DC is possible, I dare say as low as the DDS compiler allows. I guess below 1 Hz is hard due to the fractions but all above should be fine. But implementing this might take a while because there are issues with adjusting the phase and with acquiring data in the current firmware.

jbeuke, 2018-09-13:
The highest modulus seems to be 16384 resulting in a frequency resolution of 7629 Hz at the selected SFDR. If I am not mistaken, this means it's either 0 Hz or 7629 Hz or multiples of it. In standard mode we have a frequency resolution of 0.03 Hz.

jbeuke, 2020-04-24:
Not necessary anymore since the rasterized mode was dropped.

closed 2020-04-24 by jbeuke

tknopp, 2020-04-24:
yep