IBIResearch/RedPitayaDAQServer #18: About continuous write to DAC and read from ADC

liubenyuan, 2018-11-16:
Hi, thanks for this project.

Can I use `RedPitayaDAQServer` for continuous send data to DAC from PC using python (without underrun or overrun) and read samples from ADC ?

Which part should I modify FPGA/PL? I am not familiar with the development flow of redpitaya PL.

Thanks!

tknopp, 2018-11-16:
continuous send data is not possible. We have just predefined waveforms.

jbeuke, 2018-11-16:
Adding to the answer from @tknopp: Reading continuously from the ADC requires a decimation of at least eight due to bandwidth and processor speed problems. Could you maybe elaborate more on why you have to stream the data to the DAC? Maybe it is possible to just add a waveform to the existing sine, triagle, square and sawtooth signals. Is the signal periodic? Then it might be possible to add an arbitrary waveform buffer that is being read periodically.

liubenyuan, 2018-11-17:
@jbeuke Thanks for your replay. I want to send data continuously to zynq and the consumer was two slow DACs (1 MSPS), that set the positions of X-axis and Y-axis, and then use 4 ADCs to sample the feedback simultaneously. The xy position data was about ~100 Mbits, so I want to enable the continuous DAC and ADC feature in Zynq as well as the python GUI. The xy positions are not periodically and they may have arbitrary patterns.

jbeuke, 2018-11-17:
Well, I am not sure how much more the bus can handle. For the slow DACs LUT is implemented, but those only work up to 100 kHz. Is it an option to implement everything in the FPGA? What kind of control loop and signal processing do you need to do?

liubenyuan, 2018-11-18:
100 KHz (10 us) is ok for my application. I want to send two voltages to a motor (1 DAC * 2) and read the current as well as the position feedback (2 ADC * 2). The DAC update rate should be less than 10 us, and the ADC sample rate should be less than 5 us.

jbeuke, 2018-11-18:
Okay, this sounds possible using the C library and ignoring the server code. If you can relax the requirements further, it might be possible with only one RP using the slow ADCs. This spares you from implementing the communication between two devices.

closed 2019-04-27 by jbeuke