IBIResearch/RedPitayaDAQServer #19: Last working version

tknopp, 2018-11-16:
This is more for myself.

https://github.com/tknopp/RedPitayaDAQServer/commit/c55479ffdad8592696befc8916e9a771dd6a816c

is the last working version for the HH devices. The commit after that (from me) has seem to revert several things. Have not investigated further and also have not looked at the stuff added by you @jbeuke.

jbeuke, 2018-11-19:
I did not experience problems with the addition of ` && numSlowDACChan>0` in https://github.com/tknopp/RedPitayaDAQServer/commit/12c04d6fbe56dd0118e2b13c66a3e7b7e4fd2fbf. The changes afterwards add more capabilities but should not interfere with the stuff you need and might be irrelevant for you, since the signal generation apart from sine currently only works for the rasterized mode. I split up the mode setting for standard/rasterized, which means you can use the new features on one channel set to standard while enjoying the benefits of rasterized on the other channel. This might be interesting for outputting signed DC. I did not commit versions of the synthesized firmware, since I don't want too many binary files in the repository. If you want to test them, just send me a message.

jbeuke, 2018-12-15:
True, but I think none of the existing functions was changed. It was basically about adding stuff like the other waveforms and access to the DIOs.

tknopp, 2018-12-15:
no `setDACMode` was changed. 
But that is fine. One issue is that I don't know if you recent bitfile is actually working for me. I am currently using master02.bit and that was for a reason. All versions after that had issues.

jbeuke, 2018-12-15:
True, I splitted it up to allow for different modes for the two channels. I am from time to time working on it and did not notice any issues with it. What issues were you experiencing?

tknopp, 2018-12-15:
I will write you an email.

tknopp, 2020-04-10:
this is solved now.

closed 2020-04-10 by tknopp