IBIResearch/RedPitayaDAQServer #22: setting DAC amplitude with python client

amarrei, 2021-02-18:
Hello,
I have been able to test the RedPitayaDAQServer with the julia examples and this project looks like it could be a great basis for my signal analyser, which is in python using SCPI with the original RP bit-file.
I have been unable to get the example ADC.py working though. Setting and reading the channel 1 frequency works, but the amplitude and offset always reads zero and no signal on the output. I have tried with different channel and component settings (which I haven't really understood yet either) without success.
This is on a linux/ubuntu pc using python 3. 
Any ideas what needs to be changed/adapted?
Best regards,
Amir

tknopp, 2021-02-19:
Hi Amir, I need to check this, since the master branch may have some issues. Will come back to you. 

You said that you tried it with Julia and Python. Is it correct that it was working with Julia but not working with Python?

amarrei, 2021-02-19:
Hi Tobias,
Yes, I can set both DAC channels with Julia but not with Python.
Cheers,
Amir

tknopp, 2021-02-19:
Ok, this is because we usually develop only with the Julia client and then the other clients can get broken. We really want some sort of unit test at some point but its not that easy to do this automatically.

Can you have a look on the Python code yourself and look for differences? I cannot focus on this in the near time and this will get you going faster.

tknopp, 2021-02-19:
Also it would be interesting, which of the Julia programs you run. Then you can print out the commands that are send to the RedPitaya:

https://github.com/tknopp/RedPitayaDAQServer/blob/master/src/client/julia/src/RedPitayaDAQServer.jl#L32
https://github.com/tknopp/RedPitayaDAQServer/blob/master/src/client/python/redpitaya.py#L85

This helps finding the differences.

amarrei, 2021-02-23:
Thank you Tobias for pointing out where to look.
The problem was actually simple to resolve, the passing of the amplitude and offset values should be floating point and within +/-1.0V.
The line
        self.send('RP:DAC:CHannel%d:COMPonent%d:AMPlitude %d' % (channel, component, amplitude))
should be replaced with:
        self.send('RP:DAC:CHannel%d:COMPonent%d:AMPlitude **%f**' % (channel, component, amplitude))
and for the offset
        self.send('RP:DAC:CHannel%d:OFFset **%f**' % (channel, offset))

I'm still having a hard time to convert the ADC values to volts, which will be my next item after testing the DIO functions I added to the python client.
Do you do any calibration of the ADC/DAC?
Thanks for making this available ;-)

tknopp, 2021-02-24:
Ah yes, it seems that I changed the amplitude to volt at some point.

For the calibration this is indeed somewhat suboptimal. We have a more high-level package that stores the calibration values and applies them. It something like this `  value*0.00012957305 +  0.015548877`. I usually always use these two values but it would of course be better to determine them for each RedPitaya.

By the way: if you make changes to the Python code you can make a pull-request to fix this upstream.

amarrei, 2021-03-02:
Hi, I finally managed to test the I/O functions I added (at least the reading of a DI pin).
The calibration gets me closer, but I really need to do separate analog output and input calibration, even with your mentioned values I'm quite far off.
Being new to github, how do I make a pull-request? From what I understand I need to create a branch, commit to it then make the request, right? However, I cannot create my branch.

tknopp, 2021-03-02:
I have added you to the list of contributors since this makes things easier. You can then create branches directly in this repository instead of making a fork. branches can be made also from the web UI. It important that you connect your local branch to the remote one but that is described in most git tutorials much better than I can explain here.

The pull-request is then simply made using on the Github website. If you are working on a branch I can also pull in your changes.

jonschumacher, 2021-05-18:
Resolved by 5a69bfb3936bcb3d06df2059bc68f1b9eca9949b and 1e2d967d98f3857216f5794af94fc24e293f6d2a

closed 2021-05-18 by jonschumacher