IBIResearch/RedPitayaDAQServer #31: how to get .xpr(regeneration fpga)

fuerpy, 2021-05-06:
I want to add some function to the fpga part.but i can't regeneration .xpr. i want do a arbitrary wave generator.
can you update the .xpr dokument?
thank you~

tknopp, 2021-05-07:
If you execute `make_fpga_project.sh` it will generate the .xpr and all what is needed. It then will be in build/fpga/firmware

How arbitrary should your generator be? We already have one providing some common waveforms:
https://github.com/tknopp/RedPitayaDAQServer/blob/master/src/server/scpi_commands.c#L271

fuerpy, 2021-05-07:
Thank you for your reply.I have already excute make_fpga_project.sh.but it shows "command not found".I don't know which step I did wrong.Yes i have seen that part of your function.But i want to send data（waveforms） to ram to generate arbitrary signal.

tknopp, 2021-05-07:
you need to install vivado 2017.2, then you need to but the paths of vivado into you search path and then it should work.

fuerpy, 2021-05-07:
ye,i change the version in tcl.it works.thanks~

jonschumacher, 2021-05-07:
Before you start the tedious work of programming the data movement to the FPGA, you should check whether the number of samples you need for the waveform fits into the BRAM. Otherwise you need to have a similar setup as with the ADC which is streaming to the device RAM.

tknopp, 2021-05-07:
We actually have an arbitrary waveform generator for slow signals (about 1 kHz). This is what is called "SlowDAC" in the code. It can also be output on the SMA outputs.

fuerpy, 2021-05-16:
I'm sorry, i didn't find the code about "SlowDAC".and a small question also,how can I find the address in xpr,for example in your code "dac_cfg = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, mmapfd, 0x40000000);",how can I know the address and offset.thanks

nHackel, 2021-05-16:
The SlowDAC uses a few components across the system. In the FPGA image you have the ringbuffer pdm_data which is in the system block. The FPGA uses this to create the waveform. This buffer is regularly updated by the control thread, which takes values from the slowDACLUT array and writes them to pdm_data. The LUT array is filled by the client.

When you open the block design of the image you should see a tab "Address Editor" next to the "Diagram" tab. Or you can use the "Window" option at the top and select the "Address Editor" from there

fuerpy, 2021-05-16:
oh！really？！Thank you a lot! I will try to figure out how it works!

Yeah, but this is the question. when I open xpr, i can't see "Diagram",I can just see some .v dokument.I don't know why.
Thanks again for you reply.

fuerpy, 2021-05-17:
I saw the address in "Diagram"!thank you a lot.

fuerpy, 2021-05-18:
Hallo,Your Image about Clock in Block design in Doukumention is missing. And also a question How can be the output from "SlowDAC" on the SMA outputs?

fuerpy, 2021-05-20:
I check the BlockDesgin again.As I understand, if I make *((uint8_t *)(cfg)) |= 8; this command, the output will be in the SMA output?am I right?

tknopp, 2021-05-20:
Yes that is correct. Here is the code to output the SlowDAC on the SMA outputs:
https://github.com/tknopp/RedPitayaDAQServer/blob/master/src/lib/rp-daq-lib.c#L781

jonschumacher, 2022-02-09:
I think the initial question on rebuilding the image is resolved with #35 and the according changes to the documentation. The BRAM stuff might be solved in the future. I will thus link this to #35.

closed 2022-02-16 by nHackel