IBIResearch/RedPitayaDAQServer #37: Updating from new releases

jonschumacher, 2022-04-21:
We need an update path for old, but set-up RPs. If the Linux is running fine, we just have to update our specific app folder. This could/should work with local and remote zip files of the image.

tknopp, 2022-04-21:
yes, I experienced this issue as well. For me it is not even possible to update the git repository of the recent images because of issues with file permissions. Git claims that all files have a changed permission and accordingly does not let me pull the files.

jonschumacher, 2022-04-21:
Good point. I actually meant a procedure for updating RPs via network without wiping the SD card, but I experienced this Git issue as well. Will try to figure it out.

PR #40 (2022-06-10): Updating from new releases
Closes #37
closed 2022-06-10 by jonschumacher