IBIResearch/RedPitayaDAQServer #38: Installing julia on the red pitaya

killan93, 2022-05-02:
Hi,
I have an issue with the setup of the julia client library.  I have downloaded the latest julia version from the website into the RedPitayaDAQServer directory, but when I try to untar the file I get the error "tar: can't create symlink 'julia-1.7.2/lib/julia/libgit2.so' to 'libgit2.so.1.1': Operation not permitted". 
I've also tried to install julia via the package management, but it seems like there is no julia package available.

Any help on how to install julia is welcome.

jonschumacher, 2022-05-02:
I am not exactly sure what you are trying to achieve. The Julia client is not meant to be run on the RP itself. You run it from your computer and access the functions of the DAQ server (written in C and running on the RP) via network.

killan93, 2022-05-04:
Thanks for the quick response. That was an error on my side. The installation works fine now.

closed 2022-05-04 by killan93