IBIResearch/RedPitayaDAQServer #39: Connenction to Red Pitaya fails

killan93, 2022-05-04:
Hi,
I can't establish a connection to the Red Pitaya using the examples. I've copied the Alpine Linux Image (red-pitaya-alpine-3.14-armv7-20220216) to an FAT32 sd card, installed julia1.7.2 on my PC and changed the IP address in config.jl. Then I started the script (v0.4.2)  simple.jl and got
 "ERROR: LoadError: Could not connect  to rp-f09e22.local on port 5025 since the operations was timed out after 5.0 seconds!" I have double checked the IP.

I also tried to run the python script. This one gets a connection to the Red Pitaya, but fails to acquire data and stops. I have printed the send and receive command. I suspect that the ACQUISITION mode wasn't set properly.

send command RP:MODe CONFIGURATION
recieve 1
send command RP:ADC:DECimation 32
recieve 1
send command RP:TRIGger:MODe INTERNAL
recieve 1
send command RP:DAC:CH0:COMP0:FREQ 10016.025641
recieve 1
send command RP:DAC:CH0:SIGnaltype SINE
recieve 1
send command RP:DAC:CH0:COMP0:AMP 0.500000
recieve 1
send command RP:DAC:CH0:OFF 0.000000
recieve 1
send command RP:DAC:CH0:COMP0:PHA 0.000000
recieve 1
send command RP:MODe ACQUISITION
recieve 0
send command RP:MODe?
recieve CONFIGURATION
send command RP:TRIGger ON
recieve 0
send command RP:ADC:DATa:PIPELINED? 0,780,3120

nHackel, 2022-05-04:
Hello,
the release is a little bit disjointed. We are working on a new release, which should be available in the upcoming weeks (with neat new features). In the meanwhile I can try to help you get a running system. If you added the Julia package with "add", this actually pulls the latest Julia release and this can't connect to the server you currently have. 

If you want to stick with Python you can try working with the Python script and the Julia client library as a reference. Then you need to checkout commit [2f2502ec68c7b7b979b5ff17abfff4355b4b5db4](https://github.com/tknopp/RedPitayaDAQServer/commit/2f2502ec68c7b7b979b5ff17abfff4355b4b5db4) on the RedPitaya and rebuild the server with "make server".

If you want to use Julia, you can "dev" the package and then manually checkout the same commit. See [here](https://pkgdocs.julialang.org/v1/managing-packages/#developing) for documentation on dev'ing.

The python script is actually for a newer state of the server, which was not in the RedPitaya Image and there the "ACQUISITION" mode was actually still called "MEASUREMENT", this is why you got a reply of 0 with the python script

nHackel, 2022-05-15:
After which command does the receive message not work anymore?

killan93, 2022-05-16:
In simple.py, the command in line 10 is send to the red pitaya and then in RedPitayaDAQServer the script gets stuck in line 39.

nHackel, 2022-05-24:
I am unfortunately not able to find the time to reproduce your setup at the moment. I will ping you again once we release a new version with updated examples and a new complete zip archive that should work out of the box.

This should happen in the next 2 weeks

killan93, 2022-05-25:
Thanks for your reply. I totally understand that you want to focus on the new version. Looking forward to the release.

closed 2022-05-25 by killan93

nHackel, 2022-07-05:
Hello @killan93, we just released the new version of the project!