IBIResearch/RedPitayaDAQServer #46: Install documentation

406700, 2022-06-21:
I am having trouble with the install documentation, and am following your suggestion to write here. 

I can ssh into alpine linux on the device, however I get an error when trying to run the make file with make all or make server.
"Makefil:87:warning: ignoring old recipe for target 'build/fpga/xc7z020clg400-1/cores/.xpr'"  and "/bin/sh: python: not found"

FYI, https://docs.juliahub.com/RedPitayaDAQServer/w3olm/0.1.1/installation.html links to a different alpine linux image than this page..

jonschumacher, 2022-06-22:
Hi,

we are currently preparing a new release. There were some major changes both in the FPGA image and the client/server communication. This might lead to the error, since I guess you are using the version of the client from `master`. Thanks for the hint on updating the documentation. We will do so after releasing a new image for the RP.
Concerning the `git clone`: This should be not necessary anymore with the new image. An RP which is connected to the internet should then be able to do `git pull` since the Git repo is initialized and set to the appropriate remote.
In the meantime I can only recommend two things: be patient with us or build the image yourself. I know this is not really satisfactory but the main reason for this project was and still is our usage of the modified RP in pur research. This means, we sometimes "move fast and break things". We hope to finally converge to a stable API but due to the nature of our focus and the missing CI (we would need a dedicated server with a cluster of RPs which can also build the FPGA image and flash the SD cards) we can't assure that the image is working at any point in time.

I hope this helps to get the context for the moment. Apart from that I can only ask for your patience. I would love to hear whether the new release works for you in the future and what project you are working on.

Cheers
Jonas

406700, 2022-06-27:
Hi Jonas, 

Thank you for the detailed reply. I think I will take your first recommendation and wait for the code to be updated as I don't have the knowledge to build it myself. This is currently a side project in my research, and thus I can afford to wait. Do you have a rough time frame on when you expect a new release? I could likely get by with the stock image but the higher data streaming rate and client written in Julia are a big plus with this project.

I am working in optics, and hope to use the RP in a low cost detection scheme. 

Regards, 
Michael

nHackel, 2022-06-27:
Hi Michael,

we are aiming for a release in the next week. All planned features are implemented, it is just a matter of testing and updating the documentation (and finding time to do so) one last time.

Sounds like an interesting project. I'd be glad to hear from your experience with using our image when we have a new release!

Kind regards,
Niklas

406700, 2022-06-29:
Hi Jonas, 

Great to hear. I will let you know how it goes when I try the new release!

-Michael

jonschumacher, 2022-07-04:
There is now a new pre-release: https://github.com/tknopp/RedPitayaDAQServer/releases/tag/v0.5.0-rc0
Feel free to test it.

jonschumacher, 2022-07-05:
The initial question was concerning the installation description. This is now fixed in https://tknopp.github.io/RedPitayaDAQServer/dev/installation.html. I will thus close here.

closed 2022-07-05 by jonschumacher