IBIResearch/RedPitayaDAQServer #50: Problem Continous Signal Acquisition Example

skornhuber, 2022-11-12:
We are starting working with RedPitayaDAQServer for integration in our measurement systems in our high voltage lab. The hardware and the functuality is exactly perfect for our usage.
Everything of your docu went well. However I am facing some problems with Continous Signal Acquisition Example.

I start the script with the -t 3 option

(base) stefan@mbpstefan2 julia % /usr/local/bin/julia -t 3 producerConsumer.jl 
[ Info: Start Producer
[ Info: Start Consumer
(base) stefan@mbpstefan2 julia % 

if I add a @info directly after Start Consumer the info is not showing up. It seems that the script is stopping immeadeatly.

Do you have any kind of idea how to get the script running?

Thanks a lot.

Stefan.

jonschumacher, 2022-11-13:
Does simple.jl work for you?

tknopp, 2022-11-13:
That example code seems to be not  finished. One requires some ˋwait(consumer)ˋ at the end and probably some plotting code.

jonschumacher, 2022-11-13:
True, the plotting part is missing. When running in the REPL you just get a 4D matrix `buffer` which should contain your data.

tknopp, 2022-11-13:
```
(base) stefan@mbpstefan2 julia % /usr/local/bin/julia -t 3 producerConsumer.jl
[ Info: Start Producer
[ Info: Start Consumer
(base) stefan@mbpstefan2 julia %
```
This script is actually not supposed to be run this way. One should open a REPL with `julia -t 3` and then `include(producerConsumer.jl)`. And then it also makes sense to not use `wait(consumer)` because one can interact with the thread in this way.

skornhuber, 2022-11-13:
> Does `simple.jl` work for you? And I think the example purposely doesn’t block. Please try from the Julia REPL with `include`.

yes the other samples are working

skornhuber, 2022-11-13:
> ```
> (base) stefan@mbpstefan2 julia % /usr/local/bin/julia -t 3 producerConsumer.jl
> [ Info: Start Producer
> [ Info: Start Consumer
> (base) stefan@mbpstefan2 julia %
> ```
>
>
>
>
>
> This script is [...]

great exactly - I missed this instruction. Thanks a lot - now the script is working.

jonschumacher, 2022-11-13:
Awesome! I added a discrimination between interactive and non-interactive mode for this example. If you have further questions, feel free to ask.

closed 2022-11-13 by jonschumacher