IBIResearch/RedPitayaDAQServer #60: Update using the Utility function fails

jusack, 2023-04-04:
Hi, when trying to use the `update!` function to update a RedPitaya to the newest v0.6.0 release, the build process https://github.com/tknopp/RedPitayaDAQServer/blob/12b4178cc27f0a9b31cd64a948e06f5dd3c1999f/src/client/julia/src/Utility.jl#L181 of the server fails. I found the culprit to be the first command in `make server` which also fails on its own when manually connected to the RP: 
```
rp-f09290:/media/mmcblk0p1/apps/RedPitayaDAQServer# git submodule update --init
error: Your local changes to the following files would be overwritten by checkout:
	.github/workflows/main.yml
	README.md
	examples/Makefile
	examples/common-cxx/scpi-def.cpp
	examples/common-cxx/scpi-def.h
	examples/test-interactive-cxx/Makefile
	examples/test-interactive-cxx/main.cpp
	examples/test-tcp/main.c
	libscpi/inc/scpi/cc.h
	libscpi/inc/scpi/config.h
	libscpi/inc/scpi/types.h
	libscpi/src/error.c
	libscpi/src/ieee488.c
	libscpi/src/parser.c
	libscpi/test/test_parser.c
	libscpi/test/test_scpi_utils.c
Please commit your changes or stash them before you switch branches.
Aborting
Unable to checkout '224f7cee7c2c5592081e1dec640862f029323782' in submodule path 'libs/scpi-parser'
```

The whole git repository on the SD card reports a lot of unstaged changes in the permissions, which could be the cause of this issue: 
```
diff --git a/libscpi/inc/scpi/cc.h b/libscpi/inc/scpi/cc.h
old mode 100644
new mode 100755
```

I got the `make server` command to work with adding the `-f` flag to the `git submodule update --init` command. However, I am not sure if that is the correct way to go. I would be interested if someone had the same issue before!

Greetings from Lübeck!

jonschumacher, 2023-04-14:
I already removed this on master (https://github.com/tknopp/RedPitayaDAQServer/blob/44002ed504814fb5739fdc8dc6552c7161bb2b25/src/client/julia/src/RedPitayaDAQServer.jl#L373) due to this issue and since the additional dependencies resulted in longer loading times. I want to move these utilities to a subpackage but did not have time yet. I will keep this issue open as a reminder 😉

jusack, 2025-07-29:
I have added the utilities to a separate private package

closed 2025-07-29 by jusack