IBIResearch/RedPitayaDAQServer #64: Cannot acquire more than 1024 samples

Trent-Ralph, 2023-05-25:
Hi,
Thanks for this project. Looks like it will suit our needs, however I am having a couple of problems. One problem I was able to solve but I should raise here was random external triggering with V0.6. With a script just monitoring the external trigger status every few seconds it would randomly trigger ON, even with the external trigger pin shorted to GND. Installing V0.5 fixes this. 

My current issue is that I can't seem to read more than 1024 samples, when the server has to send more than one chunk. When I do the readSamples function from python client class always gets hung on the last "read away performance data" (line 69).Looking at the code it appears this is somewhat of a known issue ([here](https://github.com/tknopp/RedPitayaDAQServer/commit/c9b338c3aa11359356ebb83bac0274bd499bbc50)). I tried reverting back to the original from this commit. This stops it from hanging and allows more than 1024 samples however the limit then appears to be around 4*1024. Any extra samples returns a crazy values as shown on this plot of feeding in a 100 kHz signal. 
[image]

Just wanted to check if there is a work around for this in python or will I need to use Julia. I hoped to eventually move to labView in order to take advantage of previously written measurement protocols. 

Apologies in advanced for any errors in this. I am fairly new to all this. 
Cheers,
Trent.

nHackel, 2023-05-25:
Hello Trent,

thanks for the interest in our project! I have not noticed the trigger issue yet, but I'll look out for it. What was your setup where you observed it? Just a single RedPitaya with trigger mode set to external?

Towards the Python issue: Generally we only keep the Julia client up to date and the Python example is more to show that the server can be reached from any TCP client. So for labView or for Python I would recommend looking into the Julia code and transferring the communication into the language of your choice.

After a quick look at the Python code I think the main issue is that the Python function mixes samples and bytes. The command to the server should be given in samples. If the chunkSize is larger than the number of expectedBytes, the server is supplied with a wrong value. This is the first bug.

The second bug is that the script is receiving chunkSize bytes, when in reality it should receive chunkSize * 4. A last bug is that the script is not accounting for the last chunk to be potentially smaller if the number of requested samples is not a multiple of the chunksize. These fixes together with re-adding the "read away perf. data" line might fix all issues with the python version.

I might get around to fixing these issues in the script next week, but if you want to already try these fixes out and they work, feel free to make a pull request

Best regards,
Niklas

Trent-Ralph, 2023-05-26:
Thanks Niklas for the quick response.
Re. the trigger issue. Yeah just a single RedPitaya with the trigger mode set to external. The DIO0_P is shorted to GND.  I was just asking the RedPitaya if the trigger was on in the loop below. On V0.6 the loop would break randomly within a few seconds. With V0.5 installed I left the loop running for at least 30 mins without it breaking. I also reinstalled version 0.6V to see if it fixed the sampling size issue. I checked the trigger again and it still triggers randomly. I also ran a loop where it just prints the rp.query response and you can see a random ON response every few seconds. 
```
#Wait for trigger loop, once triggered break
print("Waiting for trigger......")
while 1:
    TR = rp.query("RP:TRIGger?")
    if TR == "ON":
        break

print("Triggered")

```
Thanks for the information on all the potential issues with the python client. I'm going to have a play around with the Julia client today to confirm that it is working fine with my RedPitaya and then I'll look into transferring the communication from to julia to python/labView.  I'll keep you updated if I have any success. 

Thanks again,
Trent.

Trent-Ralph, 2023-07-31:
I've successfully implemented the code into LabView, so this issue can be closed. Thanks for your advice. I had some trouble with the external triggering but that turned out to be due to the external trigger needing a 50 ohm terminator to work with the 50 ohm TLL source.  I'm happy to provide some LabView examples if your interested in including them in this project. I have a new problem with the decimation filter but as it unrelated to this I will open a new issue.

jonschumacher, 2023-07-31:
Good to hear! We would gladly include a PR with examples on LabView.

closed 2023-07-31 by jonschumacher