IBIResearch/RedPitayaDAQServer #66: Can get install to work on windows machine

Curatum, 2023-06-27:
Hi,

Firstly, thank you for this great project, which potentially makes my bachelor's thesis possible. 

However, I simply can't get the project to work. I formatted a new SD card to FAT32 on Linux, made it bootable, and simply copied the image on it. When I insert the SD card into my RedPitaya no LED is blinking, and it does not connect to my DHCP server on my Windows machine. The "standard" RedPitaya image, on the other hand, does connect to the DHCP server, and I can access it as usual. Did I miss something? 

Also, when I try to build the project on RedPitaya there is no root/apps folder; only the roots' folder exits, which is empty.

I hope you can help me, as I have been trying to get it working for 2 days now and don't know any further. If you need any additional information, please let me know. I have no computer science background so most things are new for me.

Thank you in advance

Max

jonschumacher, 2023-06-28:
Hi Max,

the image defaults to 192.168.1.100. Could you try setting the network adapter to something like 192.168.1.1 and then ping 192.168.1.100? If this doesn’t work, you can grab a micro USB cable and check the boot sequence with a terminal connection.

Good luck with your thesis. If you can, please share the topic. We are always interested what the project is used for.

Cheers
Jonas

jonschumacher, 2023-06-28:
Oh, the the image does not make the RP blink. We threw that out to get a leaner build process.

tknopp, 2023-06-28:
The RP image defaults to DHCP. That needs to be something different.

jonschumacher, 2023-06-28:
It should, but DHCP fails us a lot. Try the serial connection as in https://redpitaya.readthedocs.io/en/latest/developerGuide/software/console/console/console.html and check the ip.

Curatum, 2023-06-28:
Hi, 

Thank you for the fast answer. :) 

I think it's an installation problem. The serial connection only works with the standard image, not with this project. 

Here are my steps as I tried to install the image:
1. Setup of a virtual Linux environment (Debian in VirtualBox)
2. Passed through an external card reader to the virtual machine to make the SD card accessible.
3. I downloaded the project image.
4. Formatted the SD card so that it is completely empty (no other partition).
5. I created a new partition with 4 GB of storage (I also tried 8 GB).
6. Did the formatting steps (FAT32, making it bootable, etc.)
7. Created the file system
8. I copied the files onto the SD card via the GUI File Explorer (I simply unzipped the download and copied it straight to the SD).

When I now insert the SD card into Windows, it shows me that the files are on there, but the RedPitaya will not make a single life sign over serial communication.

To my thesis: 
I try to enable predictive maintenance for injection molding tools using the acoustic emission phenomenon, which needs fast continuous sampling. I hope to detect higher friction through differentiating vibrations in the ultrasonic area. ;)

Thank you for your help

Maximilian

jonschumacher, 2023-06-29:
Try to do it directly with Windows. Just format as FAT32 and copy the zip.

jonschumacher, 2023-06-30:
Glad it helped. I will close here. If you have other questions, feel free to open another issue on the topic.

closed 2023-06-30 by jonschumacher