IBIResearch/RedPitayaDAQServer #84: Available waveforms

adilorta, 2024-05-31:
Hello,

First of all, thanks for your amazing work. I am quite new in red pitaya and fpga stuff. So, I apologize if I missed something. I have a question regarding existing waveforms. I saw that sine, triangle and sawtooth options are available. I was wondering are you planning to implement arbitrary waveform or sweep sine? 

For my application, I need to use sweep sine between 1 and 100 khz coupled with triggered acquisition. I don't need high sampling rates (up to 1 MS/s/ch is sufficient). However, I need high sample numbers (100k/ch) both on waveform generation and acquisition side. I hoped that I can use your application through python to achieve this. If you think it is not possible, I am also open any suggestions/alternatives you proposed. Thanks in advance.

Kind regards,
Adil

tknopp, 2024-05-31:
Hi, on the fast channels, there is also an arbitrary waveform available. You can see that here: https://github.com/tknopp/RedPitayaDAQServer/blob/master/src/examples/julia/waveforms.jl#L52

Regarding the swept sine the question is, if you can put that all into a single chirp, or whether one changes the frequency after each measurement (leading to a lower duty cycle). I am currently not sure, how long the waveforms on the fast DAC can be. Maybe you can sketch what the sequence should look like.

adilorta, 2024-05-31:
Thank you for your fast response. I am sorry, I didn't see arbitrary waveform because it was located in julia examples. 

I am planning to use a simple sweep for my application. Sample rate for generation can be 500 khz, and my sample size is 100k (0.2 seconds). You can see what type of signal I want to use in figure below. I need to use analog input channels to record the response. Again, for acquisition, sample rate can be 500khz and sample size per channel is 100k.  

[image]

Could it possible to send this signal through your arbitrary waveform generator and acquire the measured response? I know that it is high over the limits sending 100 k signal to waveform generator and receiving 100 k from each input channel. If you think it isn't feasible, I am open to suggestions. 

For my application, I have to fix acquisition sample rate and sample number. However, if sending 100k data to waveform generator is a problem, I could be more flexible such as send my chirp signal in small pieces and record the response. Afterwards, I could combine the data with post processing. Thanks in advance.

tknopp, 2024-05-31:
Currently, the waveform is limited to 16 k, see https://github.com/tknopp/RedPitayaDAQServer/blob/e68bedaf93a142abc3810db2eefbbe88ad95d7e7/src/lib/rp-daq-lib.h#L34
So I fear that at this point, you need to chunk the waveform in pieces. But maybe @nHackel has an idea of what's possible here. It would be great if we would have some mechanism to use our "sequencing" infrastructure, which runs at about 10 kHz to feed the FPGA with new waveforms (using a double buffering strategy). Another idea would be to put amplitude/frequency in a sequence and let these values be changed on the FPGA in a synchronized fashion (per SlowDAC step).

adilorta, 2024-06-01:
Thank you so much for your help. I'll try to implement with partial waveform.

closed 2024-06-01 by adilorta