IBIResearch/RedPitayaDAQServer #87: modulus < > sample rate?????

karu2003, 2024-08-20:
Where do you get it from modulus.
I need to generate a chirp signal, I don't really understand what kind of sample rate you have. :(

tknopp, 2024-08-20:
125000000 / modulus is the sampling rate. 125Mhz is fixed in the RP.

karu2003, 2024-08-21:
it's not quite clear. how is this related to DEC? You change both values ​​in the examples.
I have another question. I'm looking for an alternative to the official firmware.
I need to get a signal of 0.3 seconds duration with a sampling frequency of at least 96000....1M at the generator output.

tknopp, 2024-08-21:
which example are you referring to? That would help.
The second question, I don't understand.

karu2003, 2024-08-21:
In all examples you don't have sampre rate. You have DEC and modulus.

Regarding the second question. Can your system reproduce a long signal. more than 16K points?
For example, I have a signal of 300K points and 1M same rate.

nHackel, 2024-08-21:
The decimation determines the sampling rate, it is based on the clock of 125 MHz. The highest sampling rate we have is achieved with a decimation of 8. That gives you 1.25e8/8 = 1.5625e7 Hz. This is essentially just for receiving the data.

The examples then use the modulus to define the frequency of the signals we are sending. They use a modulus of 12480 for example. 1.25e8/12480 is around 10 Hz. So that would be a 10 Hz sine wave. When you define your signal frequency you have to take care that you sampling rate or rather how you group your samples, fits to your signal. Otherwise you get a drift in your reception.

For example, when we use a frequency of 1.25e8/12480 and a decimation of 8 we have 1560 samples per period. With a decimation of 32 we would have 390 per period. If we would have a decimation of 128, we would have 97.5 samples per period and in the reception our signal would "drift" over time.

The [data acquisition](https://tknopp.github.io/RedPitayaDAQServer/dev/acquisition.html#Sampling-and-Data-Rates,-Transmission-Speeds-and-Time-to-Live) section of the documentation details for different sampling rates how long data exists before it is being overwritten. Generally, our system can record data continuously at the highest supported sampling rate. It is more a question of how efficient you are receiving the data. At that data rate a single RedPitaya produces 500 Mbit/s

nHackel, 2024-08-21:
I'm not exactly sure what you mean with reproduce a signal of more than 16k points. Do you mean an arbitrary waveform?

jonschumacher, 2024-08-21:
According to [this thread](https://github.com/RedPitaya/RedPitaya/issues/302), the question is about the send side and not receiving.
So actually the question about the modulus is probably just a misunderstanding. Sending is always done with 125 MHz sample rate. There is some documentation on arbitrary waveforms with LUTs in https://tknopp.github.io/RedPitayaDAQServer/dev/examples/sequence.html.

karu2003, 2024-08-21:
> I'm not exactly sure what you mean with reproduce a signal of more than 16k points. Do you mean an arbitrary waveform?

Yes. arbitrary?

jonschumacher, 2024-08-21:
https://en.wikipedia.org/wiki/Arbitrary_waveform_generator ;)

karu2003, 2024-08-21:
I have some of these. I need to replace it with something. :)

Thank you. Your firmware is no different from the original. :( 
There is also a limit of 16K points.
And DAC streaming only works at 125M.
I will look for some USB avg generator.
Thanks for introducing me to Julia.

jonschumacher, 2024-08-21:
It really depends on what you want to achieve. Could you specify the signal you want to produce?

karu2003, 2024-08-22:
simple example. sometimes i need to give an ECG signal. 
A picoscope is the worst choice. :(

    fs = 1e6  # Sampling frequency
    t1 = 0.004  # Chirp duration 4 ms
    t2 = 0.002  # Chirp duration 2 ms
    pause = 0.008  # Pause 8 ms
        
    function generate_chirp(f0, f1, t, fs)
        t = range(0, stop=t, length=Int(fs * t))
        return sin.(2 * π * (f0 .+ (f1 - f0) .* (t ./ t[end])) .* t)
    end
    function generate_signal(fs, t1, t2, pause)
        signal = Float64[]
        append!(signal, generate_chirp(18000, 34000, t1, fs))
        append!(signal, zeros(Int(fs * pause)))
        for _ in 1:2
            append!(signal, generate_chirp(34000, 18000, t1, fs))
            append!(signal, zeros(Int(fs * pause)))
        end
    
        for _ in 1:129
            append!(signal, generate_chirp(34000, 18000, t2, fs))
        end
    
        return signal
    end
    signal = generate_signal(fs, t1, t2, pause)

jonschumacher, 2024-08-22:
Hm, easy to implement on the FPGA but not so easy in our current firmware. In the end it is just a counter in front of the frequency input of a DDS combined with another counter for the pause sequencing. If the fixed sequence is sufficient, this can be done easily. Just grab the FPGA image and strip most parts. Just hardcode the values and you are good to go. But making it configurable requires implementing it in the FPGA, then the server and also the client. This is not done quickly. If you want to have a go on it, we can probably assist you but currently this is nothing we plan to implement on our own.

karu2003, 2024-08-22:
[jonschumacher](https://github.com/jonschumacher)
It's good that there are alternatives in the rack. I took ADALM2000 - 64K buffer, adjustable sample rate. The disadvantage is that it works in conjunction with a raspberry or PC. No ethernet.

jonschumacher, 2024-08-22:
Alright, seems like you found a solution. I will close here. If you need assistance in implementing a chirp within the firmware, feel free to open a new issue.

closed 2024-08-22 by jonschumacher